ArduPilot/WebTools #10: FilterReview: features to be implemented

IamPete1, 2023-06-11:
Somewhere to note down all the features that we would like:

- [ ] More speed! Faster filter calculation would allow auto-re plot.
- [ ] Show filter tracking on averaged FFT plots
- [ ] Zero pad FFT data to allow running on batch sizes that are not a power of 2. (also do some reading to check that this is mathematically valid) 
- [ ] Add plot with general flight data, (roll, pitch, throttle) to allow selection of "good" portion of flight for FFT
- [ ] Add param descriptions and bitmask helper (possibly auto generated from param meta data)
- [ ] Add more text to page describing how it works in more detail.
- [ ] Add link to filter tool for more detailed look at filter bode plots
- [ ] Support accels?

jkordani, 2025-02-18:
ESC Telem with "MultiSource" deselected doesn't respect the scale value of the _REF field.  I am working on a PR to fix that.

Additionally, I'd like to have the _REF field used as a scale for multisource as well, but that is not supported in AP firmware yet (and I'll probably make a PR for that too)

IamPete1, 2025-02-18:
> ESC Telem with "MultiSource" deselected doesn't respect the scale value of the _REF field. I am working on a PR to fix that.

I think that is what I would have expected. The way to test is to turn on notch logging and compared the estimated to the logged on the spectrograph. 

[image]

jkordani, 2025-02-18:
Not sure I understand your response, to be clear
https://github.com/ArduPilot/ardupilot/blob/15376a4908cc1a0019f692b495085688736f84e0/libraries/AP_Vehicle/AP_Vehicle.cpp#L866
If the filter has the option set for HarmonicNotchFilterParams::Options::DynamicHarmonic which in the filter review tool corresponds to "MultiSource", then individual filters are added per esc telem freqency.  If not, then the average is targeted.  In the AP code, the average is scaled by the _REF parameter, in the WebTools it is not

IamPete1, 2025-02-18:
> Not sure I understand your response, to be clear
> https://github.com/ArduPilot/ardupilot/blob/15376a4908cc1a0019f692b495085688736f84e0/libraries/AP_Vehicle/AP_Vehicle.cpp#L866

Hum, that is quite odd. As you say, it looks like were doing it wrong. I guess most everyone uses per-ESC if they have the feedback (that is what I would suggest you do too).

jkordani, 2025-02-18:
I have counter-rotating motors, so I have my first noise peak is at 1/2 the motor RPM and hence need to scale the targeted frequencies down by 1/2 of the reported RPM.  The AP doesn't support scaling on per-ESC tracking (which frankly looks like a simple oversight, I'm sure my prop style is uncommon), but it does support scaling on avg-ESC and RPM tracking.  In the short term, I we'll see how far we get with avg-ESC with scaling, but I plan on submitting a PR to add scaling on per-ESC as well.

IamPete1, 2025-02-18:
> I have counter-rotating motors, so I have my first noise peak is at 1/2 the motor RPM and hence need to scale the targeted frequencies down by 1/2 of the reported RPM.

I don't understand how it would be half?

jkordani, 2025-02-18:
I have a motor that has a pair of props on the top, and a pair on the bottom.  They rotate in opposite directions, and there seems to be a noise source at 1/2 the motor RPM.  The upper and lower blades pass each other at 1/2 motor revolution

jkordani, 2025-02-18:
Nope I'm dumb, it would be twice the frequency, good catch.