ArduPilot/WebTools #44: Filter review tool error message

criegit, 2023-09-19:
Hi,

i got a problem that the flight log is not evaluated by the tool. I just started analyzing the problem, probably something in the config mission or wrong. But the main problem is that the error message is not really helpful for the enduser. Additionally i get different errors with different browsers that relate to the same line of code. Maybe a clearer output of the error is possible? I will investigate a bit more and try different settings.

Firefox: 
TypeError: log.messages[inst_msg].time_boot_ms is undefined
URL: https://firmware.ardupilot.org/Tools/FilterReview/FilterReview.js
Line Number: 218

Chrome:
Uncaught TypeError: Cannot read properties of undefined (reading 'length')
URL: https://firmware.ardupilot.org/Tools/FilterReview/FilterReview.js
Line Number: 218

Hardware: Pix32v6/Pixhawk 6C
Ardupilot 4.3.7 with a small modification for our tiltrotor motors. but just a formula changed, no logic changes.
Log: https://gigamove.rwth-aachen.de/en/download/70274cf1bd5ee97842a5e66b057fd1f2

These are the corresponding lines of code. Line 218 is the one with "log.messages[inst_msg].time_boot_ms.length"
```
// Individual RPM
        var instances = 0
        var instance_map = []
        for (let i=0;i<16;i++) {
            const inst_msg = msg + "[" + i + "]"
            if (log.messages[inst_msg] != null) {
                this.data[instances] = { time:[], freq:[] }
                for (var j=0; j < log.messages[inst_msg].time_boot_ms.length; j++) {
                    this.data[instances].time[j] = log.messages[inst_msg].time_boot_ms[j] / 1000
                    this.data[instances].freq[j] = log.messages[inst_msg].RPM[j] / 60
                }
                instance_map[i] = instances
                instances++
            }
        }
```

IamPete1, 2023-09-19:
@criegit Thanks for the report, we have found the problem, should be fixed soon.

IamPete1, 2023-09-19:
Fix is now live, you will have to hard reload the page to make sure you have the latest version.

closed 2023-09-19 by IamPete1

criegit, 2023-09-20:
Wow. that was very quick. Thank you a lot. And also thanks for this incredible tool. It is crazy helpful!!!