ArduPilot/WebTools #60: Uncaught TypeError: Cannot read properties of undefined (Reading 'length')

Janno117, 2023-10-11:
I have created a log file with ArduCopter V4.5.0-dev (1036244c) on Matek H743 SLIM v3.

When I try to load my log, I get following error message:

"Uncaught TypeError: Cannot read properties of undefined (reading 'length')
URL: https://firmware.ardupilot.org/Tools/FilterRevies/FilterReview.js
Line Number: 1632"

I tried to reload the page, I also tried entirely different browsers to ensure the cache is clear.

Screenshot of the original error message and the log I tried to load are included below.
[image]
LOG: https://we.tl/t-gA0Gkcke29

IamPete1, 2023-10-13:
Closed by https://github.com/ArduPilot/WebTools/pull/63.

The problem is just the big gaps in the log. Once the patch goes live you can turn down the window size to get some results, but not enough that the averaging is really working. 

Your flight controller should be able to keep up with the logging rate, it might be worth seeing if a nicer SD card helps. If not batch logging is a much more manageable logging rate so you might have better luck with that.

There are some new logging options coming to AP with the `INS_RAW_LOG_OPT` which gives raw logging but only one gyro for a much reduced logging rate.

closed 2023-10-13 by IamPete1

amilcarlucas, 2023-10-14:
Thanks for fixing this.

@Janno117 please open a new report with the stuff we discussed. This one is closed now.