ArduPilot/WebTools #68: Logn.bin will not load

ben-rcs, 2023-10-15:
Hi, I have a Matek Slim H765 flight controller in a drone. This drone is ready to fly and FFT and batch are enabled. The drone has been manually tuned and I wanted to check whether the filters were in the right position. The problem I am having is that the log .bin does not want to load in the filter review tool, I keep getting the error (please try a hard reload of this page to clear its cash).

Is there a way to communicate with you? Or where I can load the log .bin file so that you can take a look at it.

thank you in advance

IamPete1, 2023-10-15:
If the hard reload does not work (ctrl + shift + r ) then upload the log to google drive or similar and drop a link here.

IamPete1, 2023-10-18:
Closing for now, @ben-rcs please re-open and upload that log if the hard reload does not fix.

closed 2023-10-18 by IamPete1