ArduPilot/WebTools #72: HardwareReport: features to be implemented

IamPete1, 2023-10-26:
Somewhere to note down all the features that we would like:

- [ ] GPS identification

SantamRC, 2024-03-22:
@IamPete1 Can you explain the other two features that needs to be added here. Maybe I can contribute

IamPete1, 2024-03-22:
> @IamPete1 Can you explain the other two features that needs to be added here. If I can contribute

I added some more detail.

SantamRC, 2024-04-15:
Hey @IamPete1 for the OSD parameters are we talking about such parameters?

```
PARM, 96586979, MSP_OSD_NCELLS, 0, 0
PARM, 96539561, OSD_TYPE, 0, 0
```
I found them in the flight logs, should we add these in checkboxes?

IamPete1, 2024-04-17:
> Hey @IamPete1 for the OSD parameters are we talking about such parameters?
> 
> ```
> PARM, 96586979, MSP_OSD_NCELLS, 0, 0
> PARM, 96539561, OSD_TYPE, 0, 0
> ```
> 
> I found them in the flight logs, should we add these in checkboxes?

We have hundreds of parameters for each element in each screen. I think a  top level one for type and then one for each screen. you can fund them all in the full list on the [wiki](https://ardupilot.org/copter/docs/parameters.html#osd-parameters):
[image]

Rasul-projects, 2026-03-09:
Hello! I am a beginner contributor interested in ArduPilot WebTools.

I noticed that this issue currently has no assignee, and I would like to contribute. If possible, I would like to start by working on a small part of this feature.

Could you please guide me on which part would be suitable for a beginner to begin with?

Thank you!