ArduPilot/WebTools #81: MAGFit: features to be implemented

IamPete1, 2023-11-30:
Somewhere to note down all the features that we would like:

- [ ] Option to enter location manually (https://github.com/ArduPilot/WebTools/pull/61#issuecomment-1805684265) 
- [ ] Use full WMM with date and altitude. https://www.ncei.noaa.gov/products/world-magnetic-model, https://www.npmjs.com/package/geomag
- [ ] Use position throughout flight rather than just using a single origin, this will make the results better for long range flights.
- [ ] Option to use GSF and GPS yaw as "truth"

Viktor74Yl, 2026-09-25:
Starting with version 4.7.0, it is possible to add initial coordinates to the parameters. 
Please add the ability to calibrate the compass without a GPS module.

j-w9, 2026-10-24:
#370 adds a Location box to MAGFit. It covers "Option to enter location manually", and the no-GPS case raised above: `AHRS_ORIGIN_LAT` / `AHRS_ORIGIN_LON` are used when the log has no `ORGN` or `POS`.