ArduPilot/WebTools #120: MagFit tools terminology and UI suggestions

rmackay9, 2024-02-01:
I was trying to use the Magfit tool and got a little confused perhaps due to terminology and layout.

In the first sentence is says, "predicted field from the World Magnetic Model"... but from then on in the web page it uses the synonym "Expected"... I guess this "Expected" is the same as "predicted"?  Maybe we should change the first paragraph to also be "expected"?

The wording in the "Expected vs measured body frame magnetic field" section is a bit confusing.  The "No motor comp" box surrounding the "Offsets", "Offsets and scale" and "Offset and iron" makes me think that checking these boxes will perhaps enable motor compensation?  .. I don't think this is what it does though.  Maybe "No motor comp" should be changed to, "Calculate Calibration with:"
[image]

IamPete1, 2024-02-04:
I have done some fixes from this in https://github.com/ArduPilot/WebTools/pull/123

> The wording in the "Expected vs measured body frame magnetic field" section is a bit confusing. The "No motor comp" box surrounding the "Offsets", "Offsets and scale" and "Offset and iron" makes me [...]

I agree its a little confusing if you have no motor compensation options. For example if there is battery current available you would get this:

[image]

Having the second set of options makes the grouping more clear I think. We only currently support two calibration options, but in the future we will probably add a a param to select which battery to use, then you will get a new set of calibrations for each battery. We could also add a "throttle" calibration set. So if we were to change the format it would have to take into account that there could be quite a lot more options in the future.

I also wanted the titles here ("No motor comp" and "Battery 1 current") to match the legend titles on the plots.

[image]

That means were limited to a short title so the legend doesn't squash the plot too much.

I'm open to better suggestions.

> For some reason I was unable to check the "Offsets and scale" and "Offsets and iron" checkboxes but there's no explanation as to why I can't.

I have added a note in the tip:
[image]

In your case you have very bad coverage, a longer flight with more orientations will give it a better chance of fitting the higher order options where it uses more parameters. 

 

> I guess there's no way to _not_ graph the Expected? It might be good to have a checked box for this as well in the renamed, "Graph" section above.

Right, you always get the expected. I'm not sure about renaming the sections. what your picking is really the calibrations. Yes we do graph them, but the last one selected is also the one that you will get when you save parameters. If we were to rename to "graph" we might then need to add another place where you select the calibration you want to save.

prathamEndu, 2024-06-13:
Another suggestion:
Please add option to graph the raw magnetometer values as well (for comparison)