ArduPilot/WebTools #130: Undefined typeError

trexgerratt, 2024-02-20:
I'm getting the following error with my logs from Arducopter 4.2

[image]

Not sure if it's an issue or just from using an older firmware version, but I was able to mitigate this issue by modifying the source (using F12 in chrome) and putting the following lines in a try/catch:
`inst[instance].freq = all[i].freq`
`inst[instance].time_ms = all[i].time`
in the catch I would simply continue the loop.

I noticed that every 4th iteration of the loop would throw this error. Any ideas what is causing this?

See attached log.
[log.zip](https://github.com/ArduPilot/WebTools/files/14340042/log.zip)

IamPete1, 2024-02-20:
Two fixes for this log. Should be live in half a hour or so, you will have to hard-reload the page. Thanks for the report!

closed 2024-02-20 by IamPete1

trexgerratt, 2024-02-20:
Beautiful. Thanks for the quick resolution!