ArduPilot/WebTools #153: Parameter transfer/copy tool

ronsand, 2024-04-24:
A nice addition the the ArduPilot Web Tools would be a tool for selecting parameters and groups of parameters when copying/transferring settings from one FC to another. There should also be a selection to keep or discard calibration type parameters (compass, gyro etc).

In any case keep up the good work. I really like the tools and I'm using them more and more :)

IamPete1, 2024-04-24:
Checkout the "Minimal Configuration" section in [Hardware Report](https://firmware.ardupilot.org/Tools/WebTools/HardwareReport/):

[image]

ronsand, 2024-04-24:
There you have it :) I missed that functionality when I looked at that tool.
Thanks 
Regards Ronny

ronsand, 2024-04-24:
One thing I see is that you will have to use a log file (.bin) to get the "minimal configuration" to show up. That means you have to make a new flight, or enable "logging disarmed", in case you make changes after the flight and want to use that as a basis for copy.
Not a big deal, but is it be possible to get the "minimal configuration" also from the .param file (it is not as of now).

Thanks again
Regards Ronny

IamPete1, 2024-04-24:
That should have worked already. It turns out we broke it for a bit.

This fixes the bug:
https://github.com/ArduPilot/WebTools/pull/155

This fixes the reason we did not notice the bug:
https://github.com/ArduPilot/WebTools/pull/154

Should be live in 30 mins or so, you will have to hard reload the page.

ronsand, 2024-04-24:
That is what i call service. Excellent work :)

closed 2024-04-24 by IamPete1