ArduPilot/WebTools #157: Improve webtools (mainly PID tool) documentation / useability

Janno117, 2024-05-14:
In the Filter review tool there's a description of what the tool does and what the tool needs in terms of logging data. If the data isn't present in the uploaded logs, the tool gives an error.

The PID-tool does neither have the text, nor the warning when there's not enough data in the logging. It just doesn't calculate the step-responses.

Proposed solution:
Post a picture of the logging bitmasks with the minimum required logging options selected at the top of each tool.
Alternatively post the parameter values of the needed logging parameter bitmasks with the minimum required options.

IamPete1, 2024-05-27:
I just have not had a chance to add that stuff yet. The step-response is a bit tricky, you can have plenty of data but still not get a result, the data also has to be good quality. 

Log bitmask values would be nice, but the values you need a different for each vehicle type.

IamPete1, 2025-07-15:
> * I checked and I have "Fast Attitude" and "Medium Attitude" both equipped, could this be of conflict?

The tool tells you the logging rate it found, on copter your looking for 400Hz.

> * We also pulled data from a 1-2 minute flight. I have trimmed the sections where we are in different flight modes, so it only pulls from the section when I'm in Acro mode.

This should not make any difference the tool should already ignore sections with nothing going on.

> * I asked a peer, and they said it could also be because we didn't initiate any steps during flight like a 3-2-1-1 or chirp maneuver to induce PID responses. How true is their statement?

Bit hard to say without seeing the log. More inputs either steps or chirps helps. There are some vids for doing flights for other PID review tools, the same types of flights should work well for this tool too.