ArduPilot/WebTools #172: The "Position offsets" 3D graph does not work in Firefox browser

xgsa, 2024-07-10:
The "Position offsets" 3D graph does not show anything in Firefox, whereas the position of the GPS module is set (note also the the axes are also not signed):
[image]

The same log in Chromium browser looks good:
[image]

Firefox version is 127.0.2 (64-bit).
Platform: Ubuntu 22.04.4 LTS.

P.S. Thank you very much for the bunch of useful tools!

IamPete1, 2024-07-10:
Thanks for the report. 

I think this is the same issue as https://github.com/plotly/plotly.js/issues/4943. I was able to reproduce by setting `privacy.resistFingerprinting` `true` in firefox `about:config`. 

Adding `https://firmware.ardupilot.org/` to `privacy.resistFingerprinting.exemptedDomains` resolved it for me (or just turning off resist finger printing of course).

[image]
 
Unfortunately I don't think there is anything we can do to fix it from our side.

closed 2024-07-10 by IamPete1

xgsa, 2024-07-11:
Thank you for the advice, but unfortunately it didn't help (I already had privacy.resistFingerprinting set to false).
I agree that it's rather a bug of plotly, still possibly you could upgrade its version if it's not the latest yet.