ArduPilot/WebTools #195: Loading large (~170MB) log: Maximum call stack size exceeded

WARGRaph, 2024-10-23:
Hi, 

my Copter Autotune gave back bad results, so I wanted to verify my filters from the log of that flight, which is ~170MB big. 
Trying to load in in Chrome gives me the following errors: 
[image]
[image]

Firefox takes a long time, but succeeds eventually. 
Clearing the cache or reloading the page or restarting the browser doesn't help.

IamPete1, 2024-10-23:
Unfortunately this is just a limitation of chrome, it has a per tab memory limit. I have done my best to minimise the memory usage, but sometimes the only way is just to use another browser. 

If you post the log I can try it here and see if its using more memory than it should, but I think its more or less expected.

WARGRaph, 2024-10-23:
I would prefer not to post it publicly, but I can send it to you if you sanitize it or delete it after checking it.

IamPete1, 2024-10-24:
> I would prefer not to post it publicly, but I can send it to you if you sanitize it or delete it after checking it.

Sure, feel free to send it privately.

IamPete1, 2024-10-25:
https://github.com/ArduPilot/WebTools/pull/197 fixes the call stack issue, but I still can't load the log on chrome. It does get a bit further and then the whole page crashes.

IamPete1, 2024-10-26:
Closed by #197

closed 2024-10-26 by IamPete1