ArduPilot/WebTools #206: max call stack exceeded

BloodSakura3774, 2024-10-30:
I was trying to see the specturgram with filter review tool, but these two logs kept causing max call stack exceeded error and I can't see the graph. Not sure what's wrong with the log but the same fc with same setting on another drone produced logs that can be used to prodece specturgram. 
logs here https://drive.google.com/drive/folders/1LtLiFE9Ib0lVgJwjcx5ni-COvWt_cCoA?usp=sharing
[image]

BloodSakura3774, 2024-10-30:
It appears to be chrome has a limit for js... Switched to firefox and it was fine.

IamPete1, 2024-11-01:
https://github.com/ArduPilot/WebTools/pull/207 gets the logs loading in chrome. The first log stil dosn't work because the ESCs report a huge RPM at the start which confuses the tool. Second log does seem to work as expected tho.