ArduPilot/WebTools #232: update_param_diff function has an implicit global variable i

AnasChebili, 2025-03-03:
In the Log Finder tool (`LogFinder.js`), in the `update_param_diff` function, the loop variable `i` is declared without `let` or `const`, making it a global variable.

Here is the relevant part of the code (line 640):

```
for (i = 1; i < rows.length; i++) {
    const param_diff = get_param_diff(rows[i].getData().info.params, rows[i-1].getData().info.params)
    rows[i].getData().param_diff = param_diff
}
```

The variable `i` should be declared with `let` to ensure proper block scoping:

```
for (let i = 1; i < rows.length; i++)
```

This is important because `i` could unintentionally be accessed or modified elsewhere, and Implicit globals can be difficult to debug. Additionally, every other loop variable in the file uses `let` so this is very inconsistent.

Would you like me to submit a PR for this? Or would you prefer it to be bundled with other small fixes?

IamPete1, 2025-03-04:
Yes a PR would be great. Thanks!

IamPete1, 2025-03-29:
close by https://github.com/ArduPilot/WebTools/pull/234

closed 2025-03-29 by IamPete1