ArduPilot/WebTools #237: # Migrate Repository to TypeScript

naman9271, 2025-03-09:
## Description  
This issue proposes migrating the entire codebase from JavaScript to TypeScript to improve type safety, developer experience, and long-term maintainability.

## Why TypeScript?
- **Type Safety**: Catch errors during development instead of runtime.
- **Better Tooling**: Enhanced IDE autocompletion, refactoring, and inline documentation.
- **Scalability**: Clear interfaces and type definitions simplify collaboration.
- **Gradual Adoption**: Existing `.js` files can coexist during migration using `allowJs`.

naman9271, 2025-03-09:
@IamPete1 Please assign me this issue so that i start working on it

IamPete1, 2025-03-09:
So far there is no build/deploy step. That extra setup will be a burden on some devs. What I would like to do as a first step is add checking to CI that could optionally be run locally. I have attempted to do this in the past but failed to find a away to get eslint to deal with HTLM directly and pull the correct includes.

Once we have checking on the existing JS we could think about moving to TypeScript one file at a time.

naman9271, 2025-03-09:
ok @IamPete1 thanks for feedback i start working on it