ArduPilot/WebTools #246: Can not read logs in Filter Review

DimaSavynskyi, 2025-04-16:
Sorry, something went wrong.

Please try a hard reload of this page to clear its cache.

If the error persists open an issue on the GitHub repo.
Include following error message:

Error: ThrottleTarget@https://firmware.ardupilot.org/Tools/WebTools/FilterReview/tracking/Throttle.js:233:28
@https://firmware.ardupilot.org/Tools/WebTools/FilterReview/FilterReview.js:2406:43
URL: https://firmware.ardupilot.org/Tools/WebTools/FilterReview/
Line Number: 678

IamPete1, 2025-04-16:
Can you share the log that cause the issue too please.

DimaSavynskyi, 2025-04-16:
[1 01-01-1970 2-00-00 am.bin.zip](https://github.com/user-attachments/files/19783744/1.01-01-1970.2-00-00.am.bin.zip)

[1 1-1-1970 2-00-00 AM.bin.zip](https://github.com/user-attachments/files/19783747/1.1-1-1970.2-00-00.AM.bin.zip)

IamPete1, 2025-04-17:
I have done a fix, should be live in 30mins or so, you will have to hard-reload the page to make sure you get the latest version.

closed 2025-04-17 by IamPete1