ArduPilot/WebTools #247: Documentation

joe-stroup, 2025-04-17:
I see from other posts that the need for documentation for some of these Web Tools is previously stated.

If it helps, I'll volunteer to write documentation - and offer examples.  But to do so, I must first be taught how all the elements of these tools work.

I'm most interested in the PID Review tool and the Filter Review tool - but I'll help out with anything.

I understand that as a volunteer effort - the developer may not be able to prioritize the time for more documentation.  More than once I've written my own software tools (years ago) and as they were only intended for my use, I didn't bother writing documentation.  

The concern from the ArduPilot community stand point is that people (e.g. Chris Rosser) may start issuing their own instructional material and it may be incomplete or incorrect.  This is bad for the community - as it detracts from the credibility of ArduPilot.

Some of you may know I have a series of technical papers published on my web site where I took the time to try to explain things I've learned that I thought might be helpful to others.  I'm considering writing a new one that reviews Chris Rosser's YouTube video on ArduPilot tuning and supplementing his material.  Of course as my knowledge on these topics is incomplete, this may be of limited value.

I've posted a comment on the ArduPilot Discuss forum and got comments that speak to one or two elements I raised.  But by and large, my main questions were unaddressed.  This makes me wonder how widely the general ArduPilot community understands the PID and Filter review tools.

I'd like to help.

IamPete1, 2025-04-17:
Any help would be much appreciated. There are really two ways to go about it.

The first is to add a page on the AP wiki for each webtool where we can have detailed explanations. 

The second would be to improve the documentation within the tool itself. For example to Filter Review tool has lots of question marks you have hover over to get more information, this has never been added to the PID tool.

From my point of view I like the built in documentation better because it is easy to keep it in sync with the tool, there is no need to update two things for a new feature. Having said that, those two tools are "feature complete" (for now anyway), I have been doing bug fixes but not new functionality.

IamPete1, 2025-04-17:
By the way, we do have a WebTools channel over on discord if you have any questions about tool, I'm more likely to see it there than on discuss (and/or @ me on discuss).

https://discord.gg/tehKpEec

Robert2101, 2026-01-05:
Hi @IamPete1, I would like to work on this.

I plan to add the missing tooltips (embedded documentation) to the PID Review tool, matching the style used in the Filter Review tool as discussed above.

I am setting up my branch now and will start with the main "Flight Data" section tooltips.

closed 2026-02-03 by IamPete1